pablorevilla-meshtastic/meshview #49: API endpoint leaks config

alecperkins, 2025-09-18:
I don't know the usecase for this endpoint, but it seems like it shouldn’t expose the entire config freely. For the most part it’s not particularly sensitive. However, some users may wish to keep the MQTT credentials or topic info private to prevent them from being used for undesired downlinking. Database URLs can also be sensitive, though less true for sqlite I suppose.

It should either redact or omit credentials, or perhaps this endpoint is only available if `DEBUG=True`.
 
https://github.com/pablorevilla-meshtastic/meshview/blob/d5272c1aecf366709d7be90ab388f85d0e16cc72/meshview/web.py#L1583-L1589

MagicalSpacePope, 2025-09-18:
Agreed. I had to block it in my nginx config.

pablorevilla-meshtastic, 2025-09-18:
@routes.get("/api/config")
async def api_config(request):
    try:
        site = CONFIG.get("site", {})
        safe_site = {
            "map_interval": site.get("map_interval", 3),        # default 3 if missing
            "firehose_interval": site.get("firehose_interval", 3)  # default 3 if missing
        }

        safe_config = {"site": safe_site}

        return web.json_response(safe_config)
    except Exception as e:
        return web.json_response({"error": str(e)}, status=500)

closed 2025-09-18 by pablorevilla-meshtastic